ome/bioformats #4110: Opening CZI compressed with ZSTD does not allow access to series (pyramid levels)

sebi06, 2023-10-20:
Dear BF Team,

- i created two CZI image datasets
- one is uncompressed the other one compressed via ZSTD (which will become the new default compression for ZEN)
- when I open the uncompressed CZI using the latest BF I get what I expect (5 pyramid levels)
- when I open the compressed CZI it opens directly while not offering the series selection (only the highest level is read)

[image]
[image]

CZI Data:

https://www.dropbox.com/scl/fi/wauh69abwh9qdmj7l7bo6/TR_M-29_uncompressd.czi?rlkey=hbdh7k906ogdcavbccqysx56l&dl=0

https://www.dropbox.com/scl/fi/igjapws4kn5x0mvkpktbb/TR_M-29_ZSTD.czi?rlkey=2cdp4gzkv614huwpndiixqsb6&dl=0

dgault, 2023-10-23:
Thanks @sebi06 for opening the issue, I can confirm that I can reproduce the issue with Bio-Formats 7.0.1, the same issue can be seen in all versions since ZSTD compression was implemented. The provided sample files have been copied to `inbox/imagesc-83733`

dgault, 2024-02-08:
Closing as resolved by https://github.com/ome/bioformats/pull/4138

closed 2024-02-08 by dgault