ome/bioformats #4118: Dont check for update if FormatTools.VERSION is "(unknown version)"

CGDogan, 2023-11-22:
Sometimes version string is unknown version https://github.com/ome/bioformats/blob/e1aaf58bee6eebe5d4fd425e2005565d42890af7/components/formats-api/src/loci/formats/FormatTools.java#L290 . in this case, print a debug log statement here https://github.com/ome/bioformats/blob/e1aaf58bee6eebe5d4fd425e2005565d42890af7/components/formats-bsd/src/loci/formats/UpgradeChecker.java#L211-L211 that cannot check update because dont know version and don't make a request that will otherwise fail and print a verbose error

sbesson, 2023-11-22:
On the principle, I agree the upgrade checker could handle the cases where `FormatTools.VERSION` is set to its fallback value and return early for instance.

Trying to understand the context, can you elaborate on the scenario you are trying to address here? `FormatTools.VERSION` should only be set to `(unknown version)` if the metadata cannot be found in the JAR manifest and should never happen with release versions.

CGDogan, 2023-11-23:
Sorry I was indeed referring to development versions - it would help save bandwidth but as you said I have never seen it happen with release versions. So this is a really, really minor issue.