ome/bioformats #4127: parseOptions strips characters after "]"

CFGrote, 2023-12-08:
Bioformats version 7.0.1

This bug report is related to ImageSc post https://forum.image.sc/t/omero-fiji-plugin-error-invalid-image-id/89470/11. 

In brief, the fiji-omero plugin reports an error (loci.formats.FormatException: Invalid image ID) when opening an image from our institute's omero. this issue was observed by some users but not by all.

Attaching jdb to a running Fiji process, I narrowed it down to the following:

Stopping in  bioformats' `loci.plugins.in.Importer.parseOptions()`, line 127
```
print arg

"location=[OMERO] open=[omero:server=ome.evolbio.mpg.de
user=grotec
port=4064
pass=XXXXX]YYYYYYY
groupID=3
iid=36540] view=Hyperstack windowless=false crop=false "
```

Note the password: It has a "]", likely closing the opening "[" after "open=".

Then, in line 132:
```print options.getId()

options.getId() = "omero:server=ome.evolbio.mpg.de
user=grotec
port=4064
pass=XXXXX"

So somewhere in between, everything after the "]" in my password must have been stripped away, including the Image ID, causing the exception in the Fiji plugin.

I hope this can be fixed.

Something else that strikes me is that passwords seem to be passed as plain text, but is this really necessary? Shouldn't the image id be all that is needed to query the image and send it to Fiji? But maybe there are other use cases where this information is actually needed.

joshmoore, 2023-12-08:
Oh wow 👏🏽

dominikl, 2023-12-11:
Thank you very much @CFGrote . I'm surprised that this is handed down to and fails at the Bioformats level, as all the authentication and transfer is done on the OMERO level.