ome/bioformats #4136: CVE-2022-1471

Sravani-K, 2023-12-27:
Hi,

We are deploying [Digital Slide Archive ](https://github.com/DigitalSlideArchive/digital_slide_archive) in our corporate environment where a security scanning tool is blocking me from deploying to production environment due to a vulnerability resolving to bioformats_package.jar. DSA uses  [large_image_wheels ](https://github.com/girder/large_image_wheels) which depends on bioformats_package.jar. 

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/SRAVAN~1.KOT/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/SRAVAN~1.KOT/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Distro | CVE ID | Compliance ID | Type | Severity | Packages | Package Version | Package License | CVSS | Fix Status | Description | Vulnerability Link | Package Path
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
ubuntu-jammy | CVE-2022-1471 | 47 | java | critical | org.yaml_snakeyaml | 1.32 | Apache License, Version 2.0 | 9.8 | fixed in 2.0 | SnakeYaml\'s Constructor() class does not restrict types which can be   instantiated during deserialization.u00a0Deserializing yaml content provided   by an attacker can lead to remote code execution. We recommend using   SnakeYaml\'s SafeConsturctor when parsing untrusted content to restrict   deserialization. We recommend upgrading to version 2.0 and beyond. | https://nvd.nist.gov/vuln/detail/CVE-2022-1471 | /opt/venv/lib/python3.9/site-packages/bioformats/jars/bioformats_package.jar

</body>

</html>

Could you please apply the fix by upgrading org.yaml_snakeyaml to version 2.0.

melissalinkert, 2023-12-27:
Bio-Formats 6.13.0 and later use snakeyaml 2.0, see https://github.com/ome/bioformats/pull/3965 and [relevant release notes](https://bio-formats.readthedocs.io/en/stable/about/whats-new.html#april).

Please let us know which version of `bioformats_package.jar` you have; if it is older than 6.13.0, you will need to upgrade your Bio-Formats to a more recent version.

Sravani-K, 2023-12-28:
> Bio-Formats 6.13.0 and later use snakeyaml 2.0, see #3965 and [relevant release notes](https://bio-formats.readthedocs.io/en/stable/about/whats-new.html#april).
>
> Please let us know which version of [...]

We have an older version of  bioformats_package.jar ,  6.11.  Will upgrade. Thanks!

closed 2023-12-28 by sbesson