ome/bioformats #4175: showinf `OutOfMemoryError` with large JPG file

glyg, 2024-03-28:
We stumble had an error on a large jpeg image  (6 624 x 40 653 px, 214 MB) when calling `showinf` (through `omero transfer prepare`)

Here is the trace:

```
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
        at java.desktop/java.awt.image.DataBufferByte.<init>(DataBufferByte.java:93)
        at java.desktop/java.awt.image.ComponentSampleModel.createDataBuffer(ComponentSampleModel.java:433)
        at java.desktop/java.awt.image.Raster.createWritableRaster(Raster.java:1121)
        at java.desktop/javax.imageio.ImageTypeSpecifier.createBufferedImage(ImageTypeSpecifier.java:1066)
        at java.desktop/javax.imageio.ImageReader.getDestination(ImageReader.java:2877)
        at java.desktop/com.sun.imageio.plugins.jpeg.JPEGImageReader.readInternal(JPEGImageReader.java:1200)
        at java.desktop/com.sun.imageio.plugins.jpeg.JPEGImageReader.read(JPEGImageReader.java:1162)
        at java.desktop/javax.imageio.ImageIO.read(ImageIO.java:1466)
        at java.desktop/javax.imageio.ImageIO.read(ImageIO.java:1363)
        at loci.formats.in.ImageIOReader.initImage(ImageIOReader.java:148)
        at loci.formats.in.ImageIOReader.initFile(ImageIOReader.java:121)
        at loci.formats.in.JPEGReader$DefaultJPEGReader.initFile(JPEGReader.java:203)
        at loci.formats.FormatReader.setId(FormatReader.java:1421)
        at loci.formats.DelegateReader.setId(DelegateReader.java:291)
        at loci.formats.in.JPEGReader.setId(JPEGReader.java:87)
        at loci.formats.ImageReader.setId(ImageReader.java:849)
        at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
        at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1035)
        at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
```
The file can be downloaded (till April 26th 2024, ping me here if you need it later) at that link:

https://filesender.renater.fr/?s=download&token=5052c708-6aeb-44b4-bae5-e570db81d985

```sh
showinf -version
Version: 6.7.0
Build date: 23 August 2021
VCS revision: aef40b42965e2fea2dfd7b431cf503c8a1980903
```

glyg, 2024-04-02:
Hi, thanks a lot @scuniff, I'm trying this today, and closing this following up

melissalinkert, 2025-05-22:
I am also able to open this file with Bio-Formats 8.1.1, using `showinf IHC45-10xLyz.jpg` when `BF_MAX_MEM` is sufficiently large. @glyg, if you are still seeing problems with the latest Bio-Formats, please feel free to re-open.

closed 2025-05-22 by melissalinkert