ome/bioformats #4189: Loading of indica tiff files

rharkes, 2024-05-14:
At the moment bioformats doesn't load indica tiff files.
The python tifffile library can do it since version 2023.12.9.

The problem with the files is that they have 32 bit per pixel, but didn't set the TIFFTAG_SAMPLEFORMAT, so it defaults to UINT. However, it has to be SAMPLEFORMAT_IEEEFP. 

See also:
https://github.com/cgohlke/tifffile/blob/master/tifffile/tifffile.py#L8369  (the sampleformat is changed here)
https://github.com/cgohlke/tifffile/blob/master/tifffile/tifffile.py#L10459  (this is how it is checked)

dgault, 2024-05-15:
Thanks @rharkes for reporting the issue. Are there any sample Indica Tiff files that could be used for testing? If you have some suitable samples and need an upload location then we recommend using https://zenodo.org/

cgohlke, 2024-05-15:
A sample file was posted at [forum.image.sc](https://forum.image.sc/t/trouble-generating-ome-tiffs-in-the-right-shape/89191/4). See also the [comments](https://forum.image.sc/t/trouble-generating-ome-tiffs-in-the-right-shape/89191/12) regarding the file format.