ome/bioformats #4199: LIF Reader: index out of bounds

jburel, 2024-06-19:
```
java.lang.ArrayIndexOutOfBoundsException: 32
	at loci.formats.in.LIFReader.initFile(LIFReader.java:587)
	at loci.formats.FormatReader.setId(FormatReader.java:1443)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
	at ome.formats.importer.OMEROWrapper$4.setId(OMEROWrapper.java:167)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelFiller.setId(ChannelFiller.java:234)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:293)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.Memoizer.setId(Memoizer.java:662)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at ome.formats.importer.ImportCandidates.singleFile(ImportCandidates.java:427)

```

See https://www.openmicroscopy.org/qa2/qa/feedback/41723/

melissalinkert, 2024-06-19:
A file was not submitted for QA 41723, correct? We can probably prevent this specific exception, but it may indicate a file that has other issues (which we wouldn't be able to test without the file itself).

jburel, 2024-06-19:
No file submitted, I requested it

PR #4222 (2024-09-02): LIF: fix offset calculation for certain datasets with multiple unstitched files
Fixes #3873, fixes #4199.

See files in `inbox/qa/31243` and `inbox/gh-3873`. Without these changes, `showinf` should throw an `ArrayIndexOutOfBoundsException` as described in the issues. With these changes, both files should initialize and display images without exception.

ImageJ may be the easiest way to test, as much of this data has small pixel values, so the images may be hard to check with just `showinf`. `Composite` color mode may also be useful to more closely match what LAS X shows. Screenshots of LAS X f [... cut]
closed 2024-09-02 by sbesson