ome/bioformats #4216: loci.formats.FormatException while converting series to tiff format

de9821, 2024-08-01:
**my current version bftools is below**

> Version: 7.3.1
> Build date: 11 July 2024
> VCS revision: 41e12f630ad7349caa0293cba23948c17d22732b

**I am getting below error while converting vsi file series into tiff**

> ../bftools/bfconvert -no-upgrade  -bigtiff  -series 1  -pyramid-scale 1 -noflat ./DATA_022/DATA_022.vsi ./DATA_022_1.tiff
> ./DATA_022/DATA_022.vsi
> CellSensReader initializing ./DATA_022/DATA_022.vsi
> [CellSens VSI] -> ./DATA_022_1.tiff [Tagged Image File Format]
> Tile size = 512 x 512
> 	Converted 1/1 planes (100%)
> Tile size = 512 x 512
> Exception in thread "main" loci.formats.FormatException: (w:512 + x:3584) must be <= 4034
> 	at loci.formats.FormatWriter.checkParams(FormatWriter.java:469)
> 	at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:223)
> 	at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:469)
> 	at loci.formats.ImageWriter.saveBytes(ImageWriter.java:260)
> 	at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:1076)
> 	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:908)
> 	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:841)
> 	at loci.formats.tools.ImageConverter.main(ImageConverter.java:1318)
	
**Series description is below**

> 	Image count = 1
> 	RGB = true (3) 
> 	Interleaved = true
> 	Indexed = false (false color)
> 	Width = 4034
> 	Height = 4710
> 	SizeZ = 1
> 	SizeT = 1
> 	SizeC = 3 (effectively 1)
> 	Tile size = 512 x 512
> 	Thumbnail size = 109 x 128
> 	Endianness = motorola (big)
> 	Dimension order = XYCZT (uncertain)
> 	Pixel type = uint8
> 	Valid bits per pixel = 8
> 	Metadata complete = true
> 	Thumbnail series = false
> 	-----
> 	Plane #0 <=> Z 0, C 0, T 0

I referred [documentation](https://bio-formats.readthedocs.io/en/stable/users/comlinetools/conversion.html#cmdoption-bfconvert-pyramid-scale) and tried various options but still getting error. 

Also tried older version of 6.3.1 as it was suggested in old discussion in [forum](https://forum.image.sc/t/bfconvert-error-with-vsi-file/39534) but no luck.

please suggest if you faced similar issue and have any solution

dgault, 2024-08-06:
Hi @de9821, thank you for reporting the issue. If you leave out the `-pyramid-scale 1` option, do you still see the same error?

de9821, 2024-08-07:
Hi @dgault thanks for the response. 

I see similar error with out `-pyramid-scale 1` option. 

> ../bftools/bfconvert -no-upgrade -bigtiff -series 1  -noflat ./DATA_022/DATA_022.vsi ./DATA_022_1.tiff
> ./DATA_022/DATA_022.vsi
> CellSensReader initializing ./DATA_022/DATA_022.vsi
> [CellSens VSI] -> ./DATA_022_1.tiff [Tagged Image File Format]
> Tile size = 512 x 512
> 	Converted 1/1 planes (100%)
> Tile size = 512 x 512
> Exception in thread "main" loci.formats.FormatException: (w:512 + x:3584) must be <= 4034
> 	at loci.formats.FormatWriter.checkParams(FormatWriter.java:469)
> 	at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:223)
> 	at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:469)
> 	at loci.formats.ImageWriter.saveBytes(ImageWriter.java:260)
> 	at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:1076)
> 	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:908)
> 	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:841)
> 	at loci.formats.tools.ImageConverter.main(ImageConverter.java:1318)

dgault, 2024-08-09:
Thanks @de9821 for confirming. Would you be able to share a sample file showing the error that we can test? If you need a suitable upload location then you can upload files to the [Bio-Formats Zenodo community](https://zenodo.org/communities/bio-formats).