ome/bioformats #4227: Nikon TIFF channel metadata

melissalinkert, 2024-09-04:
@DavidStirling raised that Nikon TIFF channels are not handled as nicely as they could be, particularly when importing into OMERO.

For `curated/nikon/carol` in particular (and likely others), the fact that the channels are treated as 3 RGB channels and modelled as a single OME-XML `Channel` with 3 `SamplesPerPixel` means that the same emission/excitation wavelengths are applied to all 3 channels. On import to OMERO, this means that all 3 channels have the same color by default.

Original metadata indicates that there are different emission wavelengths recorded for each channel, so the better behavior would be to split the channels in the reader so that 3 OME-XML `Channel`s each with 1 `SamplesPerPixel` are recorded.

gerring, 2025-10-07:
Is this still an issue? Do you have an example file David? If not I may have some Nikon files to test with.

melissalinkert, 2025-10-07:
Yes, this is still an issue, and yes, we do have relevant example data as indicated in the issue description (it just isn't publicly available). If you have additional data that reproduces this problem, please feel free to upload publicly to https://zenodo.org/communities/bio-formats/records and link here.