ome/bioformats #4239: Empty image.name returned by `showinf` for some Zeiss .czi file

technics3, 2024-09-12:
I opened an issue to `omero-cli-transfer` because of a trouble i got with some .czi file : https://github.com/ome/omero-cli-transfer/issues/87

Here is the dataset i used if you want to run some test with it (expiring 12/10/2024) : https://filesender.renater.fr/?s=download&token=dd096ce7-4f49-4c37-a23e-620e1106842f

[@glyg](https://github.com/glyg)

joshmoore, 2024-09-12:
Can you please upload the dataset to https://zenodo.org/communities/bio-formats/records?q=&l=list&p=1&s=10 so that a license and DOI can be clearly assigned? Thanks!

melissalinkert, 2025-03-17:
Image names may legitimately be empty, so we do need the file uploaded to https://zenodo.org/communities/bio-formats/ in order to investigate. Closing, but please feel free to re-open if the file is uploaded to Zenodo.

closed 2025-03-17 by melissalinkert