ome/bioformats #4262: Error using bfGetReader - Java.lang.IndexOutOfBoundsException

rotemyamshahor, 2024-12-22:
Hello 
I've been trying to run a MATLAB script that uses bioformats to convert ets/vsi images taken by Olympus vs200 to tiff & png files. 
I'm getting this error when trying to read the vsi:

>> bfr = BioformatsImage('_00.vsi')
log4j:WARN No appenders could be found for logger (loci.formats.ClassList).
log4j:WARN Please initialize the log4j system properly.
Error using bfGetReader (line 85)
Java exception occurred:
java.lang.IndexOutOfBoundsException: Index: 6, Size: 6
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at loci.formats.in.CellSensReader.parseETSFile(CellSensReader.java:1198)
	at loci.formats.in.CellSensReader.initFile(CellSensReader.java:734)
	at loci.formats.FormatReader.setId(FormatReader.java:1421)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelFiller.setId(ChannelFiller.java:223)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:293)

Error in BioformatsImage/getReader (line 711)
                obj.bfReader = bfGetReader(obj.filename);

Error in BioformatsImage/set.filename (line 175)
            obj = obj.getReader;

Error in BioformatsImage (line 130)
                    obj.filename = varargin{1};

Could you please help with this? I've looked at issue #4117 but couldn't understand the solution. 
Thanks!

melissalinkert, 2024-12-24:
@rotemyamshahor, which version of Bio-Formats are you using? If you have something other than 8.0.1, does upgrading to 8.0.1 make a difference?

melissalinkert, 2025-03-06:
As #4117 was expected to fix errors like this, closing in the absence of version information. If this error still occurs with Bio-Formats 8.1.1 please feel free to re-open and we'll investigate further.

closed 2025-03-06 by melissalinkert