ome/bioformats #4264: .vmic reader

kwiechen, 2025-01-16:
Hello,

before making a PR of: https://forum.image.sc/t/vmic-whole-slide-image-reader/106485/3

I am using code from https://github.com/usnistgov/pyramidio to read the deepzoom image pyramid. Where should these files go?

Best regards,

Kai

melissalinkert, 2025-01-22:
Thank you for asking this, @kwiechen.

Please do not copy in files from pyramidio; instead, add it as a dependency in the appropriate pom.xml file. The dependency declaration is provided by pyramidio:

https://github.com/usnistgov/pyramidio?tab=readme-ov-file#maven-dependency

but replace the `...` with the correct version number. This declaration should go in `components/formats-bsd/pom.xml` if your reader is added under `components/formats-bsd/` (and thus BSD-licensed), or in `components/formats-gpl/pom.xml` if your reader is added under `components/formats-gpl/` (and thus GPL-licensed).

Before opening a pull request with a new reader, please make sure to read the following:

- https://ome-contributing.readthedocs.io/en/latest/third-party-policy.html
- https://forum.image.sc/t/bio-formats-update/101734
- [A successful pull request that adds a new reader](https://github.com/ome/bioformats/pull/4255)

As all of those pages indicate, we will do our best to evaluate and include your pull request. However, if the proposed changes need substantial work, or are too complex to review and test in a reasonable amount of time, we will let you know and close the PR. This is not to discourage you from opening a pull request, but rather to be clear that we may not have the capacity to spend substantial time on this new reader. If that turns out to be the case, we’ll do our best to provide suggestions for how to proceed before closing.

kwiechen, 2025-01-22:
Thank you for the information, @melissalinkert 

The signed Contributor License Agreement was sent to contributors@openmicroscopy.org

An example .vmic dataset was uploaded to Zenodo.org

The .vmic wsi files (https://openslide-users.andrew.cmu.narkive.com/f3mFAKMI/precipoint-vmic-driver-and-deepzoom) are nested zip archives with a deepzoom image pyramid. In order to read pixel data and metadata without unzipping, I have used the java.nio.file.FileSystem. In addition .vmic files may be generated in sparse scan mode. Therefore the pyramidio deepzoom reader was (slightly) modified to work with the zip file system provider and missing image data.

Best regards,

Kai

melissalinkert, 2025-01-22:
As a place to start then, how about:

- push a branch with your changes to pyramidio
- open a draft pull request against this repository with your .vmic reader in its current state
- link to your pyramidio branch from the draft pull request
- provide a direct link to the data uploaded to Zenodo

Then we can see the exact scope of changes, and provide better suggestions accordingly. Without being able to reference your code, it's a little difficult to make specific recommendations.

kwiechen, 2025-01-23:
The reader in its current state uses 4 files from pyramidio. Parts of this may be integrated or removed, especially if no .vmic or deepzoom writer functions are planned. https://github.com/usnistgov/pyramidio was not maintained for 9 years. How about first updating my bioformats fork with the .vmic reader and providing the example data link?

kwiechen, 2025-01-25:
vmic reader: https://github.com/kwiechen/bioformats-vmic
example data: https://zenodo.org/records/14714280