ome/bioformats #4281: Leica (.LIF) files importing colors incorrectly

arelirdza, 2025-02-26:
When I import a .lif image, the channels are imported in the wrong order. It appears that the red channel and blue channel are swapped. I have had this issue with both brightfield and polarized images.

Attached is an example of the color discrepancy. The image as it should appear is shown in the Leica LASX window, and the image as it is loaded with Bio-Formats is shown in the QuPath window.

[image]
[image]

melissalinkert, 2025-02-26:
Thanks for reporting this, @arelirdza. Could you please upload a .lif file that reproduces this problem to https://zenodo.org/communities/bio-formats? The screenshots are clearly different, but without the original .lif file it will be difficult to test a fix.

This may be related to #3655, but we would need the .lif file to confirm.

melissalinkert, 2025-02-26:
Thanks for uploading; the file is in https://zenodo.org/records/14933318.