ome/bioformats #4310: Incorrect handling of -omexml-only errors

AlexBarbera, 2025-05-16:
[This line](https://github.com/ome/bioformats/blob/32a5c82137daa5709ea6679b2e9c896da193a87b/components/bio-formats-tools/src/loci/formats/tools/ImageInfo.java#L233) makes some errors not be displayed depending on the position of the arguments.

ie. Notice that `-no-pix` is an incorrect flag and therefore should error
`showinf -no-pix -omexml-only image.ome.tif` <- Will display "Invalid parameter -no-pix"
`showinf -omexml-only -no-pix image.ome.tif` <- Will error but without any message

Perhaps the output should be redirected to STDERR or somewhere else. Disabling it completely can lead to confusing situations like the one mentioned above.
Another potential solution is redefining the flag, where the output is omexml only if no errors are found.

Perhaps a note on the documentation is an alternative too?

Thanks,
Alex

melissalinkert, 2025-06-06:
Thanks, @AlexBarbera. #4323 is one fairly straightforward way to fix this - if you're interested in trying that out, any feedback would be welcome.

closed 2025-06-12 by sbesson

AlexBarbera, 2025-06-12:
Well done and thanks for the quick fix, i ran a quick test and it is now consistent with the expected outputs.
👍 
I wonder if is worth discussing that completely shutting down the logger is perhaps a bit extreme. It could be worth looking into adding a "omexmlonly" level that still allows errors to be displayed.

commit 6885243d (2025-06-12): showinf: configure logging for `-omexml-only` after all arguments parsed

Fixes #4310.