ome/bioformats #4312: how to read qptiff format tiled image?

chengdonglin, 2025-05-25:
qptiff have multi channels , how to use the lib to read target channel tiled image ? have any example ?

sbesson, 2025-05-29:
Yes the first series will be the whole slide image in this example. 
To access the data at a particular pyramidal level, you will first need to call [reader.setResolution(level)](https://downloads.openmicroscopy.org/bio-formats/8.2.0/api/loci/formats/IPyramidHandler.html#setResolution-int-) where the level 0 corresponds to the highest resolution and level n-1 corresponds to the lowest resolution before using `openBytes` to read the tile at the desired level for a given channel.

chengdonglin, 2025-06-01:
> Yes the first series will be the whole slide image in this example. To access the data at a particular pyramidal level, you will first need to call [...]

Thank you very much—your information has been extremely helpful to me.

closed 2025-06-01 by chengdonglin