ome/bioformats #4313: Error parsing schema

caylamason, 2025-05-29:
Hi there,

I am using bftools version 8.2.0. I attempted to validate the XML after successfully converting a czi file to ome.tiff and received the following error:

$bftools/xmlvalid lab_processed/images/sample.ome.tiff
Parsing schema path
http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
Validating lab_processed/images/3041_pB_F.ome.tiff
Error parsing schema at http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
org.xml.sax.SAXParseException: s4s-elt-character: Non-whitespace characters are not allowed in schema elements other than 'xs:appinfo' and 'xs:documentation'. Saw '301 Moved Permanently'.
  	at org.apache.xerces.util.ErrorHandlerWrapper.createSAXParseException(Unknown Source)
	at org.apache.xerces.util.ErrorHandlerWrapper.error(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.xs.opti.SchemaDOMParser.characters(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanContent(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl$FragmentContentDispatcher.dispatch(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanDocument(Unknown Source)
	at org.apache.xerces.impl.xs.opti.SchemaParsingConfig.parse(Unknown Source)
	at org.apache.xerces.impl.xs.opti.SchemaParsingConfig.parse(Unknown Source)
	at org.apache.xerces.impl.xs.opti.SchemaDOMParser.parse(Unknown Source)
	at org.apache.xerces.impl.xs.traversers.XSDHandler.getSchemaDocument(Unknown Source)
	at org.apache.xerces.impl.xs.traversers.XSDHandler.parseSchema(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaLoader.loadSchema(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaLoader.loadGrammar(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaLoader.loadGrammar(Unknown Source)
	at org.apache.xerces.jaxp.validation.XMLSchemaFactory.newSchema(Unknown Source)
	at javax.xml.validation.SchemaFactory.newSchema(SchemaFactory.java:638)
	at javax.xml.validation.SchemaFactory.newSchema(SchemaFactory.java:670)
	at loci.common.xml.XMLTools.validateXML(XMLTools.java:871)
	at loci.common.xml.XMLTools.validateXML(XMLTools.java:785)
	at loci.formats.tools.XMLValidate.validate(XMLValidate.java:67)
	at loci.formats.tools.XMLValidate.validate(XMLValidate.java:104)
	at loci.formats.tools.XMLValidate.main(XMLValidate.java:125)

Based on the error message, it seems to be an issue with the schema formatting. Would you mind taking a look?

Thank you,
Cayla

sbesson, 2025-05-30:
@caylamason thanks for opening this issue. I can easily reproduce using Bio-Formats 8.2.0 and any of the official OME-XML or OME-TIFF samples. More specifically, the problem is not specific to Bio-Formats 8.2.0 and can be reproduced using the command-line utility for any Bio-Formats version.

The issue comes from the fact each file stores a reference to the OME XSD schema using http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd as per the specification. An HTTP -> HTTPS 301 redirect has been recently introduced at the level of http://www.openmicroscopy.org recently and some of the Java tooling fails to handle these redirects:

```
sbesson@Sebastiens-MacBook-Pro-3 Downloads % curl -IL http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
HTTP/1.1 301 Moved Permanently
Server: nginx/1.28.0
Date: Fri, 30 May 2025 13:31:17 GMT
Content-Type: text/html
Content-Length: 169
Connection: keep-alive
Location: https://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd

HTTP/1.1 200 OK
Server: nginx/1.28.0
Date: Fri, 30 May 2025 13:31:17 GMT
Content-Type: application/octet-stream
Content-Length: 261500
Last-Modified: Wed, 28 May 2025 14:48:42 GMT
Connection: keep-alive
ETag: "6837224a-3fd7c"
Accept-Ranges: bytes
```

At the code level, there are a few possibilties to mitigate this issue:
- update the low-level validation tools to support 301 redirects
- resurrect #3268 which was a previous attempt to use of the cached XSD schemas instead of making HTTP(S) requests. Note this is already the strategy used when calling `showinf -omexml`

@jburel @pwalczysko could you comment on the infrastructure changes that has been made at the level of the OME resources?

joshmoore, 2025-06-18:
:+1: for the fix of xmlvalid in in #4316, but do we know if the issue will effect others?

jburel, 2025-06-30:
Prior to change
```
 tools/xmlvalid ~/Desktop/course_lif/output.ome.tiff
Parsing schema path
http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
Validating /Users/jmarie/Desktop/course_lif/output.ome.tiff
Error parsing schema at http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
org.xml.sax.SAXParseException: s4s-elt-character: Non-whitespace characters are not allowed in schema elements other than 'xs:appinfo' and 'xs:documentation'. Saw '301 Moved Permanently'.

```
After the change in the nginx configuration
```
tools/xmlvalid ~/Desktop/course_lif/output.ome.tiff
Parsing schema path
http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
Validating /Users/jmarie/Desktop/course_lif/output.ome.tiff
No validation errors found.

```

Let me know if any issue